dotintent/flutter-nfc-reader #1: test on genuine android device

matteocrippa, 2018-07-20:
I have no Android device capable to read NFC so I wasn't able to debug and check my "blind" implementation, it would be great to test that a little on real device and debug if needed

deetu, 2018-07-23:
Sure @matteocrippa, added but its as if its silently failing.

matteocrippa, 2018-07-23:
@deetu nothing was prompt to ask you for permission when opening the app, isn't it? I mean no alert with request for permission?

deetu, 2018-07-23:
@matteocrippa Yea, I wasn't prompted, I think the above lines in the manifest already took care of that. Also, like I said the version and stop functions returned results and when I turned off the NFC, I got a result about the unavailability of NFC functionality, so I guess all should be fine except for the actual reading.

matteocrippa, 2018-07-23:
@deetu can you try adding these lines in you AndroidManifest in your demo app?

```xml
<intent-filter>
            <action android:name="android.nfc.action.NDEF_DISCOVERED"/>  
            <data android:mimeType="mime/type" />
        </intent-filter>
```

deetu, 2018-07-23:
@matteocrippa I tried that earlier using 'text' as the mineType. With the last suggestion, I'm stilling getting the same thing. 
'W/permissions(12182): [Ljava.lang.String;@95b2d5e' Thats what i get.

vinicentus, 2018-07-23:
I don't think one should get a permission popup for nfc? Other nfc apps that I have used doesn't do that, apparently android handles it as a minor permission.

deetu, 2018-07-23:
@matteocrippa You are right about that, NFC is grouped under PROTECTION_NORMAL. I will keep checking and if you are able to fix it soon too, do let me know.

matteocrippa, 2018-07-23:
I'm cleaning right now all the stuff and updating to latest libs and Kotlin version, so the project is more clean inside.

Then I think the issue is connected to something else, I need to move the logic in 

```kotlin
override fun onNewIntent(intent: Intent?) {
        super.onNewIntent(intent)
        if (intent == null) {
            return
        }
```

I just need to figure out what flutter library I need to add to be able to override that in the code.

vinicentus, 2018-07-23:
Error when building:
`\flutter-nfc-reader\android\src\main\kotlin\com\example\flutternfcreader\FlutterNfcReaderPlugin.kt: (16, 1): Class 'FlutterNfcReaderPlugin' is not abstract and does not implement abstract member public abstract fun onActivityResult(p0: Int, p1: Int, p2: Intent!): Boolean defined in io.flutter.plugin.common.PluginRegistry.ActivityResultListener`

deetu, 2018-07-23:
@matteocrippa Will give it a short and update you

deetu, 2018-07-23:
@matteocrippa Tried that, it's not showing any error, but from the look of things the 'read' method isn't really been called, so it returns nothing, i.e Like I never called the function. The other two functions were tested and they returned results.

vinicentus, 2018-07-24:
When detecting `ACTION_TECH_DISCOVERED`, shouldn't there be an intent filter in the example AndroidManifest.xml? Also, shouldn't there be a list of supported technologies then is a xml file? https://developer.android.com/guide/topics/connectivity/nfc/nfc#tech-disc
https://github.com/matteocrippa/flutter-nfc-reader/blob/afe4b1bfc140dcc57b462cec8ef4e6aee3406b23/android/src/main/kotlin/com/example/flutternfcreader/FlutterNfcReaderPlugin.kt#L73

matteocrippa, 2018-07-24:
`ACTION_TECH_DISCOVERED` is a kinda fallback for not _NDEF_ formatted, prob is better atm to stay focus only on NDEF, have that working fine and then expand it

nyrf, 2018-07-30:
Not work on my android phone,  `FlutterNfcReader.read;` seems return null.

matteocrippa, 2018-07-30:
I pushed a new test if @nyrf or @deetu what to try it, it would be awesome

deetu, 2018-07-30:
Hey @matteocrippa, will give it a try tonight and update you. Thanks

nyrf, 2018-07-31:
@matteocrippa Thanks, but it still not work for me, i tried to use react native with 
https://github.com/whitedogg13/react-native-nfc-manager and it works.

matteocrippa, 2018-07-31:
I need to get back an Android phone, debugging without it is kinda complicated.

I tried pushing a patch and updating the instruction to use that, I would ask if you can test using directly the example and see what happens with debugger on

nyrf, 2018-07-31:
I've tested and it show nothing. i used vsc to code.

```
Launching lib/main.dart on NX531J in debug mode...
Built build/app/outputs/apk/debug/app-debug.apk.
W/System  (12015): in loadDexFile: /system/framework/QPerformance.jar
```

matteocrippa, 2018-07-31:
Mhh prob is better to use Android Studio opening directly the android example there, using vsc and using flutter will not allow to get all the information.

My only concern is that the `activity` is not get in proper way

nyrf, 2018-07-31:
The log

```
07-31 19:01:07.055 30154-24807/? E/BrcmNfcJni: getReconnectState = 0x0
07-31 19:01:07.059 30154-30154/? E/NfcService: applyRouting -20.
07-31 19:01:07.060 30154-24807/? E/BrcmNfcJni: getReconnectState = 0x0
07-31 19:01:07.060 30154-30154/? E/BrcmNfcJni: nfcManager_disableDiscovery:UICC_LISTEN_MASK=0x07;
07-31 19:01:07.060 30154-30154/? E/BrcmNfcJni: nfcManager_disableDiscovery:P2P_LISTEN_MASK=0x0196;
07-31 19:01:07.060 30154-24807/? E/BrcmNfcNfa: UICC/ESE[0x402] is not activated
```

```
07-31 19:01:06.093 3195-3274/? D/ZLauncher3: [LauncherModel] PackageUpdatedTask.run, op=2 ,packages=[Ljava.lang.String;@f121bf0  mIsRepeatVerifyCount == 0
    [LauncherModel] OP_UPDATE com.android.nfc
07-31 19:01:07.051 30154-30154/? D/BrcmNfcJni: nfcManager_doSetScreenState: state = 1
    acquireRfInterfaceMutexLock: try to acquire lock
    acquireRfInterfaceMutexLock: sRfInterfaceMutex lock
    startRfDiscovery: is start=0
07-31 19:01:07.051 30154-30154/? I/BrcmNfcNfa: NFA_StopRfDiscovery ()
07-31 19:01:07.051 30154-24807/? I/BrcmNfcNfa: NFA got event 0x010F
    nfa_dm_evt_hdlr event: NFA_DM_API_STOP_RF_DISCOVERY_EVT (0x0f)
    nfa_dm_act_stop_rf_discovery ()
    nfa_dm_rf_deactivate () deactivate_type:0x0
    nfa_dm_disc_sm_execute (): state: DISCOVERY (1), event: DEACTIVATE_CMD(6) disc_flags: 0x2
    NFC_Deactivate 4 (IDLE) deactivate_type:0
    nfc_ncif_send_cmd.
07-31 19:01:07.051 30154-24807/? D/NfcAdaptation: NfcAdaptation::HalWrite
07-31 19:01:07.051 30154-24807/? D/NxpExtns: const nxp::CNfcParam *nxp::CNfcConfig::find(const char *) const found MIFARE_READER_ENABLE=(0x1)
07-31 19:01:07.054 30154-24807/? I/BrcmNfcNfa: nfa_dm_disc_sm_execute (): new state: DISCOVERY (1), disc_flags: 0x22
    nfa_sys_ptim_stop_timer a5d253e8
    ptim timer stop
    Stopped presence check timer (if started)
07-31 19:01:07.055 30154-24812/? D/NfcAdaptation: NfcAdaptation::HalDeviceContextDataCallback: len=4
07-31 19:01:07.055 30154-24807/? I/BrcmNfcNfa: NFC received rsp gid:1
    P2P_Prio_Logic
    nfa_dm_p2p_prio_logic type = 0xff
07-31 19:01:07.055 30154-24807/? I/BrcmNfcNfa: returning TRUE
07-31 19:01:07.055 30154-24807/? I/BrcmNfcNfa: nfc_set_state 4 (IDLE)->4 (IDLE)
    nfa_dm_disc_data_cback ()
    nfa_dm_disc_discovery_cback (): event:0x4005
    nfa_dm_disc_sm_execute (): state: DISCOVERY (1), event: DEACTIVATE_RSP(7) disc_flags: 0x22
    nfa_dm_disc_new_state (): old_state: DISCOVERY (1), new_state: IDLE (0) disc_flags: 0x2
07-31 19:01:07.055 30154-24807/? D/BrcmNfcJni: nfaConnectionCallback: event= 31
    nfaConnectionCallback: NFA_RF_DISCOVERY_STOPPED_EVT: status = 0
    notifyPollingEventwhileNfcOff: sDiscCmdwhleNfcOff=0
07-31 19:01:07.055 30154-24807/? E/BrcmNfcJni: getReconnectState = 0x0
07-31 19:01:07.055 30154-24807/? I/BrcmNfcNfa: nfa_dm_start_rf_discover ()
    NFA_DisablePolling ()
07-31 19:01:07.055 30154-30154/? I/BrcmNfcNfa: NFA_DisablePolling ()
07-31 19:01:07.056 30154-24807/? I/BrcmNfcNfa: NFA got event 0x0107
    nfa_dm_evt_hdlr event: NFA_DM_API_DISABLE_POLLING_EVT (0x07)
    nfa_dm_act_disable_polling ()
    nfa_dm_deactivate_polling ()
    nfa_dm_delete_rf_discover () handle=0x2
07-31 19:01:07.056 30154-24807/? D/BrcmNfcJni: nfaConnectionCallback: event= 1
    nfaConnectionCallback: NFA_POLL_DISABLED_EVT: status = 0
07-31 19:01:07.056 30154-30154/? D/BrcmNfcJni: SetScreenState: enter
    Set Screen OFF
    SetScreenState: Success NFA_SendNxpNciCommand
07-31 19:01:07.056 30154-24807/? I/BrcmNfcNfa: NFA got event 0x011A
    nfa_dm_evt_hdlr event: Unknown or Vendor Specific (0x1a)
07-31 19:01:07.056 30154-24807/? D/NfcAdaptation: NfcAdaptation::HalWrite
07-31 19:01:07.056 30154-24807/? D/NxpExtns: const nxp::CNfcParam *nxp::CNfcConfig::find(const char *) const found MIFARE_READER_ENABLE=(0x1)
07-31 19:01:07.057 30154-24812/? D/NfcAdaptation: NfcAdaptation::HalDeviceContextDataCallback: len=4
07-31 19:01:07.057 30154-24807/? D/BrcmNfcJni: NxpResponse_SetDhlf_Cb Received length data = 0x4 status = 0x0
07-31 19:01:07.058 30154-30154/? D/BrcmNfcJni: releaseRfInterfaceMutexLock: sRfInterfaceMutex unlock
07-31 19:01:07.059 30154-30154/? E/NfcService: applyRouting -20.
07-31 19:01:07.059 30154-30154/? D/NfcService: applyRouting - enter force = true mScreenState = 1
    computeDiscoveryParameters Reader_ON .............
07-31 19:01:07.059 30154-30154/? D/BrcmNfcJni: nfcManager_disableDiscovery: enter;
[... 95 lines cut ...]
    NFC_DiscoveryStart
    NFC_DiscoveryStart status: 0xe3
07-31 19:01:07.065 30154-24812/? D/NfcAdaptation: NfcAdaptation::HalDeviceContextDataCallback: len=5
07-31 19:01:07.065 30154-24807/? I/BrcmNfcNfa: NFC received rsp gid:0
    nci_proc_core_rsp opcode:0x2
    nfa_dm_nfc_response_cback () NFC_SET_CONFIG_REVT(0x5002)
07-31 19:01:07.065 30154-24807/? D/NfcAdaptation: NfcAdaptation::HalPrediscover
07-31 19:01:07.065 30154-24807/? I/BrcmNfcNfa: nfc_ncif_send_cmd.
07-31 19:01:07.065 30154-24807/? D/NfcAdaptation: NfcAdaptation::HalWrite
07-31 19:01:07.065 30154-24807/? D/NxpExtns: const nxp::CNfcParam *nxp::CNfcConfig::find(const char *) const found MIFARE_READER_ENABLE=(0x1)
07-31 19:01:07.067 30154-24812/? D/NfcAdaptation: NfcAdaptation::HalDeviceContextDataCallback: len=4
07-31 19:01:07.067 30154-24807/? I/BrcmNfcNfa: NFC received rsp gid:1
07-31 19:01:07.067 30154-24807/? I/BrcmNfcNfa: returning TRUE
07-31 19:01:07.067 30154-24807/? I/BrcmNfcNfa: nfa_dm_disc_discovery_cback (): event:0x4000
[... cut ...]
    nfaConnectionCallback: NFA_RF_DISCOVERY_STARTED_EVT: status = 0
07-31 19:01:07.067 30154-24807/? I/BrcmNfcNfa: nfa_dm_disc_sm_execute (): new state: DISCOVERY (1), disc_flags: 0x1
07-31 19:01:07.067 30154-30154/? D/BrcmNfcJni: SecureElement::resetRfFieldStatus`: enter;
    SecureElement::resetRfFieldStatus`: exit
    releaseRfInterfaceMutexLock: sRfInterfaceMutex unlock
    nfcManager_disableDiscovery: exit
07-31 19:01:07.097 3195-3274/? I/ZLauncher3: [LauncherModel] start mOp ==> 2 com.android.nfc mUser UserHandle{10}
```

matteocrippa, 2018-07-31:
seems like it receives something early than expected and turn off the NFC pooling in the intent

deetu, 2018-08-01:
@matteocrippa Been doing some research, i think the issue is with this line 
    private val nfcAdapter: NfcAdapter? by lazy { NfcAdapter.getDefaultAdapter(activity) }

It returns an error, this probably will be because it's been called before the functionality is ready. Im new to Kotlin, but will read about how to create something similar to 'oncreate' in android.

matteocrippa, 2018-08-01:
@deetu can you copy/paste the error?
I pushed an updated moving the setup in a listener of onCreate, if you want to test it

deetu, 2018-08-01:
@matteocrippa Testing out the last commit seems to be working. I will test it rigorously and update you soon.

nyrf, 2018-08-02:
I got the same message: `NFC Hardware not found.`

matteocrippa, 2018-08-02:
ok there's something that is not clear, so the listeners I was expecting to work doesn't.
So it seems it doesn't fire up the NFC reader and it turns to be `null`.

I tried a different approach, and applied a patch too for simulator only (to avoid crash) due to there's no NFC support on that.

nyrf, 2018-08-02:
Seems to return null again.

matteocrippa, 2018-08-02:
@nyrf  But it raises the `NFC Hardware not found.` message? or not?

nyrf, 2018-08-02:
@matteocrippa No, show nothing.

matteocrippa, 2018-08-02:
this turn to be strange, because if it doesn't show that, it means it is not able to get the NFC instance from Android OS.
I can try changing from `context` to `activecontext` provided by flutter, but dunno if it makes sense or fix

nyrf, 2018-08-03:
I have updated to use the master branch, but it still show nothing. seems `FlutterNfcReader.read` return null.
[image]

matteocrippa, 2018-08-03:
@nyrf mhh but till now what do you used so far? In order to avoid to issue releases I always pushed to master only, so there's the risk I updated something and you were testing somewhere else? :)

In order to test git clone the master repo, open `example` project of android with android studio and run that only, don't use flutter directly.

I'm trying to figure out how does it return null, in Kotlin there's a waterfall of test to avoid that null...

deetu, 2018-08-03:
I have tested the app and using the master branch and attached herewith is the screenshot
[image]

matteocrippa, 2018-08-03:
NFC will be my biggest mystery on Android :(
Btw I tried to write from scratch the plugin using a different approach, cross your finger meanwhile testing

nyrf, 2018-08-09:
Any updated?

matteocrippa, 2018-08-09:
I've released 6 days ago.a new version on `master`, but not received any feedback yet

deetu, 2018-08-09:
@matteocrippa I have tried the new code. In the 'startNFC' function the return type isn't a string but a Uint8ArrayView, which needs to be converted to a String, this causes an error in the flutter_nfc_reader.dart file as the Future return type is expected to be a string. Once this is resolved, I think the only other issue will be with the stopNFC function, doesn't seem to stop it, but that can be resolved later.

matteocrippa, 2018-08-09:
I will look into that but not earlier than a couple of weeks, if anyone wants to help any PR is welcome

nyrf, 2018-08-09:
I've tried and still return null.

matteocrippa, 2018-08-09:
@nyrf seems weird, because @deetu have a different return object, are you sure that you are testing the latest master commit?

deetu, 2018-08-10:
`
import android.os.Bundle;
import android.app.PendingIntent;
import android.nfc.NdefMessage;
import android.nfc.NfcAdapter;
import android.nfc.Tag;
import android.nfc.tech.Ndef;
import android.os.Parcelable;
import android.content.Intent;
import android.util.Log;
import android.content.IntentFilter;

import io.flutter.app.FlutterActivity;
import io.flutter.plugins.GeneratedPluginRegistrant;
import io.flutter.plugin.common.MethodCall;
import io.flutter.plugin.common.MethodChannel;
import io.flutter.plugin.common.MethodChannel.MethodCallHandler;
import io.flutter.plugin.common.MethodChannel.Result;

import java.io.UnsupportedEncodingException;

public class MainActivity extends FlutterActivity {
    NfcAdapter nfcAdapter;
    String nfcData;
    private static final String CHANNEL = "addyourchannelidentifier";
    PendingIntent mPendingIntent;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        GeneratedPluginRegistrant.registerWith(this);
        nfcAdapter = NfcAdapter.getDefaultAdapter(this);

        new MethodChannel(getFlutterView(), CHANNEL).setMethodCallHandler(
                new MethodCallHandler() {
                    @Override
                    public void onMethodCall(MethodCall call, Result result) {
                        nfcData = null;
                        if (call.method.equals("nfcRead") && nfcAdapter != null) {
                            final Intent intent = getIntent();
                            if (nfcAdapter.isEnabled()) {
                                nfcData = getNfcData(intent);
                                result.success(nfcData);
                                nfcData = "";
                            } else {
                                result.success("NFC is not enable");
                            }
                        } else {
                            result.success("Unsupported Phone");
                        }
                    }
                });
    }

    public void onPause() {
        super.onPause();
        nfcAdapter.disableForegroundDispatch(this);
    }

    private String getNfcData(Intent intent) {
        return readFromIntent(intent);
    }

    private String readFromIntent(Intent intent) {
        String action = intent.getAction();
        String ret = "";
        if (NfcAdapter.ACTION_TAG_DISCOVERED.equals(action)
                || NfcAdapter.ACTION_TECH_DISCOVERED.equals(action)
                || NfcAdapter.ACTION_NDEF_DISCOVERED.equals(action)) {
            Parcelable[] rawMsgs = intent.getParcelableArrayExtra(NfcAdapter.EXTRA_NDEF_MESSAGES);
            NdefMessage[] msgs = null;
            if (rawMsgs != null) {
                msgs = new NdefMessage[rawMsgs.length];
                for (int i = 0; i < rawMsgs.length; i++) {
                    msgs[i] = (NdefMessage) rawMsgs[i];
                }
            }
            ret = buildText(msgs);
        }
        return ret;
    }

    private String buildText(NdefMessage[] msgs) {
        if (msgs == null || msgs.length == 0)
            return null;
        String text = "";
        byte[] payload = msgs[0].getRecords()[0].getPayload();
        String textEncoding = ((payload[0] & 128) == 0) ? "UTF-8" : "UTF-16"; // Get the Text Encoding
        int languageCodeLength = payload[0] & 0063; // Get the Language Code, e.g. "en"
        try {
            text = new String(payload, languageCodeLength + 1, payload.length - languageCodeLength - 1, textEncoding);
        } catch (UnsupportedEncodingException e) {
            Log.e("UnsupportedEncoding", e.toString());
        }
        return text;

    }

}
`

I have been able to get a working code above in my MainActivity.java file, I get the right results. I only need to refactor and probably solve the issue that the app restarts whenever I place a tag close to it. My end point will be that the app remains opens, for a better user experience when the tag is placed close by and scanned. I think I'm just missing something, but I will figure it out sometime today.

nyrf, 2018-08-10:
@matteocrippa Yes, i used the master branch and debug with vscode, return null.
[image]

matteocrippa, 2018-08-10:
@nyrf VSCode at this stage is not the optimal way to proceed, Android Studio running only the example helps more because you have full logs from device

@deetu I will look into that, I edited you comment to fix the code visualisation it was broken

deetu, 2018-08-10:
@matteocrippa Thanks, I was almost sleeping off when I posted it. Yeah, have a look and see how you can edit your plugin, I think you most likely will just need the buildText functionality in yours like I said earlier.  I have also taken out unused imports, which were added while I was testing.

matteocrippa, 2018-08-10:
I quickly look into that, but you are using a different approach, the one I used for most of the time without any success, now I switched to a new one that use readTag, prob I got rid of the byte array stuff, if I have 10 mins later I will check and push a fix

gomiunik, 2018-11-04:
Hi. 
I've tested it directly by adding package to pubspec.yaml. I needed to remove the app from the device and after reinstalling it (with another debug session) the NFC reader works.

Is there any way to change the default system sound?

Tested it on Android Kitkat

matteocrippa, 2018-11-04:
@gomiunik seems not https://stackoverflow.com/questions/8743588/prevent-alert-tone-when-scanning-identifying-an-nfc-intent

gomiunik, 2018-11-05:
@matteocrippa Have you already tried to set the FLAG_READER_NO_PLATFORM_SOUNDS
(https://stackoverflow.com/a/21308366/107076) ?
I'm still a noob on this, but your plugin is a great help

closed 2018-12-08 by matteocrippa