dotintent/flutter-nfc-reader #3: NFC Status shows true when nfc is off

vinicentus, 2018-07-23:
When startNFC runs, and if all before the setstate fails, the _nfcActive will still be set to true, even though that doesn't mean nfc is present and active.
https://github.com/matteocrippa/flutter-nfc-reader/blob/master/example/lib/main.dart#L46-L62

matteocrippa, 2018-07-23:
you are right I will clean up that too, I want to prioritise the Android stuff first and later clean this up for both the two archs

matteocrippa, 2018-07-23:
@vinicentus updated the example in the meanwhile, it should be more clear.
Also I noticed I made a mistake before, now fixed in latest version.

```dart
Future<Null> startNFC() async {
    String response;
    bool state = false;

    try {
      final String result = await FlutterNfcReader.read;
      if (result == null) {
        response = '';
      } else {
        response = result;
      }
    } on PlatformException {
      response = '';
      state = false;
    }
    setState(() {
      _nfcActive = true;
      _nfcData = response;
    });
  }
```

closed 2018-07-24 by vinicentus