dotintent/flutter-nfc-reader #6: Atm only Swift ? what is required to do in configuration ?

fvisticot, 2018-11-02:
What is required on XCode project side to allow 'Atm only Swift' ?

Without any special configuration I have the following error:
```

=== BUILD TARGET connectivity OF PROJECT Pods WITH CONFIGURATION Debug ===
    /Users/xxx/.pub-cache/git/flutter-nfc-reader-e4c0dfd361993826decaf9c526622e0bd47ed29d/ios/Classes/FlutterNfcReaderPlugin.m:2:9: fatal error: 'flutter_nfc_reader/flutter_nfc_reader-Swift.h' file not found
    #import <flutter_nfc_reader/flutter_nfc_reader-Swift.h>
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

matteocrippa, 2018-11-02:
The project needs to be created on flutter with swift flag to work

fvisticot, 2018-11-02:
> The project needs to be created on flutter with swift flag to work

Can you please provide the way  to update the Android Studio Flutter projet to set this "swift flag" ?

matteocrippa, 2018-11-02:
You can't update, you have to create the app from scratch  in this way

```bash
 flutter create -I swift nameofproject
```

fvisticot, 2018-11-02:
> You can't update, you have to create the app from scratch in this way
> 
> ```shell
>  flutter create -i swift nameofproject
> ```

Why did you add this constraint ? Is it because the plugin is coded in SWIFT ?
If i fork the projet and do my objc implementation it will work ?

matteocrippa, 2018-11-02:
For my purposes swift was enough, no need to keep compatibility over objc, so neither never dig why it's not working. Either if it was supposed to.
Dunno if a fork will do the job, btw you can even create a swift project and use objc inside...

closed 2018-11-02 by matteocrippa