dotintent/flutter-nfc-reader #9: Android - Response is the Tag Id/SerialNumber rather than the content

AcidNoX, 2018-12-07:
Testing this on a Android device and the response appears to be the Tag Serial Number / Id rather than the content:

[image]
[image]
[image]

As you can see the hex code returned matches the serial number rather than the text content on the card (which is "Hello").

matteocrippa, 2018-12-08:
I made a patch, but had no way to test that, try using the master version and see if it works

abhisheklal04, 2018-12-10:
The latest master code doesn't works.

CosimoSguanci, 2018-12-16:
I'm also facing this issue. It returns the serial number of the NFC Tag

matteocrippa, 2018-12-16:
Give a try to latest in master after merge if a pr

CosimoSguanci, 2018-12-16:
How can I use that? Now i've ^0.0.21 in my pubspec.yaml file.

matteocrippa, 2018-12-17:
about to push`0.0.22` hope to do later this morning

CosimoSguanci, 2018-12-17:
I tried the latest version. Now, if the text written in the tag is 'flutter!!!' I get 
`�Tenflutter!!!!` by reading it.

kgamecarter, 2018-12-30:
I need the ID.
Can response a object with content and ID?

matteocrippa, 2019-01-04:
@CosimoSguanci @kgamecarter give a try to this branch feature/refactor

kgamecarter, 2019-01-05:
There is no responses when I read Mifare Classic card.
I only need Mifare Classic card ID.

matteocrippa, 2019-01-05:
Try using debug to see if the callback has been triggered, so we can figure out how to proceed or if the callback is the bottleneck.

I have no mifare tag so no way to test by myself

Schillen, 2019-01-09:
using refactor branch: after reading an nfc tag type "nxp mifare classic" the app displays

Status: NFCStatus.none
Identifier : questionmark-icon + random letters
Content:
Error:

gavinerasmus, 2019-01-12:
I receive the NFC ID on a Samsung Galaxy S6 when using version 0.21. I created a new project, copied across the code and used v0.22 and got no response from the reader. I'm going to do some investigating and see what I can figure our and will let you know. v0.21 is great - thanks for creating it!

matteocrippa, 2019-02-07:
up is now the `master` working fine?

CosimoSguanci, 2019-02-07:
For me yes, it works except from the empty ID that i get on iOS (i opened another issue)

crzdg, 2019-02-08:
Hi there,

I'm using the package since today. I was able to set up a reader in my scoped_model based app. I'm also get the my cards content. However, the information in the bytes before the payload is broken. My message should be:

urn:nfc:ext:app.io:IDid12345

The string I get is somehow broken and only show the payload properly. It would have been easy for me to split the string and extract the values out of it, as I'm defining my NDEFMessages. 

However, I used the NFC technology now heavily in one of my projects. I do have some basic knowledge about NFC and NDEF. A nice documented NFCReader project with is implementing and explaining the NDEF Format is found here:

https://nfcpy.readthedocs.io/en/latest/
NDEF Docu: https://nfcpy.readthedocs.io/en/latest/topics/ndef.html

About the formats. The NDEF Format is as you may know the open NFC format. If you want to read seperate formats you have to respectively handle the byte code. If you want to get more information feel free to contact me for a skype call or somewhat.

crzdg, 2019-02-08:
String is somehow false encode. Would be nice if it would encode after NDEF Format. Good resources are found here.

https://nfcpy.readthedocs.io/en/latest/topics/ndef.html
https://nfcpy.readthedocs.io/en/latest/overview.html

matteocrippa, 2019-02-09:
@crzdg create a new dedicated issue, if you can provide also some debug info (eg. expected result, result obtained it could help more.

I will close this because it seems to work now for id/serial, if issue is still open, re-open the task

closed 2019-02-09 by matteocrippa