dotintent/flutter-nfc-reader #10: startNfc() doesn't begin reading NFC.

RandyStevens, 2018-12-09:
Test Device: SamsungS7Edge

Flutter: Flutter 1.0.1-pre.3

[image]

Attached below is screenshot of device. I switched RaisedButton for FlatButton hoping it was an error with RaisedButton
[image]

matteocrippa, 2018-12-09:
Need further detail to investigate, also because I have no android device to test.

- Did you follow the android setup ( https://github.com/matteocrippa/flutter-nfc-reader#android-setup )?
- Which version of the lib are you using?

RandyStevens, 2018-12-09:
The most recent version of the lib. I Just cloned the repo, then from example I built, then: 
`pub install && flutter run`

From what I can see the example app has the "android setup" already done, so I haven't messed with AndroidManifest.xml

What other details are relevant and can help? Thanks for the fast response.

matteocrippa, 2018-12-09:
Which kinda of NFC are you using? NDEF one?
btw, give a try to this instead of `master` https://github.com/matteocrippa/flutter-nfc-reader/releases/tag/0.0.21

RandyStevens, 2018-12-09:
Gave the tagged release you linked a try. Exact same results. 

Yes. I am using NDEF
```
           <intent-filter>
                <action android:name="android.nfc.action.NDEF_DISCOVERED"/>
                <category android:name="android.intent.category.DEFAULT"/>
                <data android:mimeType="text/plain" />
            </intent-filter>
```

matteocrippa, 2018-12-09:
The tagged one was working for other dev, but returning only the id of the NFC tag, so there's something more that need to be debugged, dunno if Samsung device has any extra requirements/grant to allow NFC reading.
But debugging that for me is a bit complicated right now

RandyStevens, 2018-12-09:
Ill try to do what i can :)

matteocrippa, 2018-12-17:
Refer to #9 it should be fixed

closed 2018-12-17 by matteocrippa