dotintent/flutter-nfc-reader #11: NDEF URL cut off on iOS

prullmann, 2018-12-14:
When reading tags on iOS with flutter-nfc-reader, I noticed that often the URLs get cut off and only the last part is returned to my code.
Example: FlutterNfcReader.read returns "itter.com/caseyneistat/status/1073422967718187010" instead of "https://twitter.com/caseyneistat/status/1073422967718187010"

It does not seem consistent when and how much is cut off and I found indications it is different between iPhone 7 and 8. I still have to verify that though.

Maybe it is some form of buffering issue?
Or did I overlook something in using the library?
Does this behaviour ring a bell? Does someone have an indication on where to look?

Thanks in advance,
Peter

prullmann, 2018-12-14:
The first 3 bytes of the NDEF payload is cut off by the current implementation. I found [example code](https://www.appcoda.com/corenfc-introduction/) doing the same to cut off a leading language code.

But it looks to me that (since iOS 12?) this language code is not there anymore, so removing the first three bytes will just cripple the actual content/URL.

I don't know what is the best way to detect and remove this language code in a more intelligent way. Since I don't have a device with iOS 11 and I can't test NFC in the emulator, I don't know how this language code looks like.

The inconsistency I mentioned was caused by different NDEF content types and flutter-nfc-reader not respecting type and typeNameFormat, just returning payload. But this is another issue.

matteocrippa, 2018-12-17:
I look into that, need to think if it’s in scope or not to decode too or read only the content and forward as it is

matteocrippa, 2019-01-04:
@prullmann did you have time to test the branch I prepared for you?

prullmann, 2019-01-06:
@matteocrippa excuse my late reply.
Yes, I tested and it works for me. As I said, I don't know however what happens on iOS11, where there are apparently three extra bytes of language code sent at the start of the payload.
I made my code resilient to this by looking for the hostname I am interested in and not caring for anything before that, but it may perhaps cause troubles for others.

closed 2019-01-09 by matteocrippa