dotintent/flutter-nfc-reader #15: Cannot get example working

RafeAldridge, 2019-01-03:
I can get the example to run on my android phone, but pressing Start NFC does nothing- Screen says: "Running on: Android 7.0" "NFC is Reading: false" "NFC Data: "

I used a native app to confirm that the NFC tags I have are readable on this device. 

Any thoughts? 

Rafe

```
Launching lib/main.dart on SM G390F in debug mode...
Initializing gradle...
Resolving dependencies...
Gradle task 'assembleDebug'...
Built build/app/outputs/apk/debug/app-debug.apk.
Installing build/app/outputs/apk/app.apk...
Syncing files to device SM G390F...
D/ViewRootImpl@76c46cb[MainActivity](30710): ViewPostImeInputStage processPointer 0
W/System  (30710): ClassLoader referenced unknown path: /system/framework/QPerformance.jar
E/BoostFramework(30710): BoostFramework() : Exception_1 = java.lang.ClassNotFoundException: Didn't find class "com.qualcomm.qti.Performance" on path: DexPathList[[],nativeLibraryDirectories=[/system/lib, /vendor/lib]]
V/BoostFramework(30710): BoostFramework() : mPerf = null
D/ViewRootImpl@76c46cb[MainActivity](30710): ViewPostImeInputStage processPointer 1
D/SemPersonaManager(30710): isNFCAllowed
D/ViewRootImpl@76c46cb[MainActivity](30710): ViewPostImeInputStage processPointer 0
D/ViewRootImpl@76c46cb[MainActivity](30710): ViewPostImeInputStage processPointer 1
D/SemPersonaManager(30710): isNFCAllowed
```

matteocrippa, 2019-01-03:
branch `master`?

If so there's something strange with android.
Btw as far as I can see if it says that reading is always `false` somehow is not triggering the reading loop

RafeAldridge, 2019-01-03:
Thanks for the quick reply. 

Sorry, I forgot to say; yes, branch master. Though I've also tried refactor and that doesn't work either - that also has NFC is Reading: False, always.

Any thoughts as to what I can try?

matteocrippa, 2019-01-03:
Hard to say, I can’t neither debug. I’m an iOS dev usually 😅

RafeAldridge, 2019-01-03:
No worries. I'm only playing around so it's not overly important.

We'll both have to wait until a more experienced Android/Flutter person comes along ;)

Appreciate your help.

matteocrippa, 2019-01-04:
@RafeAldridge which flutter version are you using?
Can I ask you to try installing using dart package, the last stable should be 0.22 or something as far as I remember and it was working for other devs

RafeAldridge, 2019-01-04:
@matteocrippa Flutter 1.0.0 

Have tried using the dart package rather than a local copy. Have also tried two different Android devices (Android 6.0.1 and Android 7.0). 

Both are exactly the same - nothing happens when I press the "Start NFC" button

matteocrippa, 2019-01-04:
Just double check if permission are granted, because as you can see just a couple of issue down it works for other ppl

matteocrippa, 2019-01-04:
I advise you to try using the app with Android Studio and debug and see what's happening.

At the very end the code for flutter, once you tap the button will always turn read on and off if it recognise the tag or fails

```dart
String response;
    bool reading = true;

    try {
      response = await FlutterNfcReader.read;
      reading = false;
    } on PlatformException {
      response = '';
      reading = false;
    }
    setState(() {
      _nfcReading = reading;
      _nfcData = response;
    });
```

Lidgett, 2019-01-06:
I have the exact same issues as @RafeAldridge.

NFC reading is always false, nothing happens.  I have tried on Android 7.0 and 8.0, physical devices of course.

I haven't tried the iOS version yet, I'm hoping to test it soon as it's an important part of my app development.  Would I have to have the Flutter package supporting Swift or will Objective C work too with the plugin?

matteocrippa, 2019-01-06:
@Lidgett 
Swift is better, and Android should work on Kotlin, at least plugins are built in that languages.

The behaviour you are both facing is pretty strange, I do suggest debug the Android app and see what's triggered by the plugin and what not.

As you can see on #9 it works

@RafeAldridge @Lidgett 
Btw I have just release a new fix under `feature/refactor`, give a try to that.

Lidgett, 2019-01-06:
Thanks for the prompt reply.  I've tried the "feature/refactor" release.  Same issue.  The code never goes beyond the invokeMethod("NfcRead") method.  So presumably there's something happening from within the Android code.  
I'll try to debug it at some point.

matteocrippa, 2019-01-06:
@Lidgett which version of Android is running on the device you are using? < Kitkat?

Lidgett, 2019-01-06:
Currently 8.0

matteocrippa, 2019-01-06:
Seems more a permission issue, I tried adding something, but dunno if makes any progress

RafeAldridge, 2019-01-07:
@matteocrippa really appreciate the help, but still no joy on Android 7.0 or 6.01. Both devices show the attached but neither reads a tag. 

Apologies, but I'm totally new to flutter and Android development. Am just playing around with a proof of concept. So my debugging skills in this area aren't much use. 

[image]

RafeAldridge, 2019-01-07:
If it helps, this is the results from NFC Tools app re the tag I'm using.

[image]

Schillen, 2019-01-08:
i have the same issue on virutal ios and real android 9. if i check the permission of the app there is nothing set. it vibrates on contatct with an nfc chip, but does not update the state / widget screen does not change.

Schillen, 2019-01-08:
yes, simulator.

i set the needed permission in the android manifest. should i check something else as well?

was working with master, gonna test the refactor branch next and let you know

matteocrippa, 2019-01-08:
Simular for iOS should not work, at the end it has not NFC reader so it will always get stuck there.

For Android, better the refactor branch, I've added there a new approach and a way to ask permission for Android >= 6.0

Schillen, 2019-01-08:
in the refactor branch(https://github.com/matteocrippa/flutter-nfc-reader/tree/feature/refactor) the plugin version is flutter_nfc_reader: ^0.0.23

which throws this error 
/Users/schillen/development/flutter/bin/flutter --no-color packages get
Running "flutter packages get" in flutter_app_nfc_refactor...     
Because flutter_app_nfc_refactor depends on flutter_nfc_reader ^0.0.23 which doesn't match any versions, version solving failed.

pub get failed (1)
Process finished with exit code 1

matteocrippa, 2019-01-08:
you have to link to that as GitHub repo and not using version, it's not worth to release 0.0.23 till is stable.

```yml
flutter_nfc_reader:
    git:
      url: git://github.com/matteocrippa/flutter-nfc-reader.git
      path: feature/refactor
```

Schillen, 2019-01-09:
"flutter packages get" throws the following error

Running "flutter packages get" in flutter_app_nfc
Could not find a file named "feature/refactor/pubspec.yaml" in git://github.com/matteocrippa/flutter-nfc-reader.git f3ca502b1995f3629b6c9b995227e5a1d6b12cdc.

pub get failed (1)

matteocrippa, 2019-01-09:
@Schillen I fixed above, it should work now

RafeAldridge, 2019-01-09:
Hi, 

Apologies, I was away from my desk the last couple of days so unable to test. 

Have now tried with the latest in feature/refactor and get an error (below). I've tried with ../ and 

flutter_nfc_reader:
    git:
      url: git://github.com/matteocrippa/flutter-nfc-reader.git
      ref: feature/refactor

in my pubspec.yaml

Running "flutter packages get" in example...                     0.5s
Launching lib/main.dart on SM G390F in debug mode...
Initializing gradle...
Resolving dependencies...
Gradle task 'assembleDebug'...
e: /Users/rafe/Development/flutter/.pub-cache/git/flutter-nfc-reader-cbcac8cb701d9d61f8e38a8b85698f0bfc9d229c/android/src/main/kotlin/it/matteocrippa/flutternfcreader/FlutterNfcReaderPlugin.kt: (115, 25): Only safe (?.) or non-null asserted (!!.) calls are allowed on a nullable receiver of type ByteArray?

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_nfc_reader:compileDebugKotlin'.
> Compilation error. See log for more details

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s
Finished with error: Gradle task assembleDebug failed with exit code 1

RafeAldridge, 2019-01-09:
We need to get you an Android phone! I'd consider sponsoring this if you're interested?

matteocrippa, 2019-01-09:
For device I have, but at home, not where I'm currently. Have just to get that next time.

I pushed a little tweak, it should fix the optional value now

matteocrippa, 2019-01-09:
@RafeAldridge try now please

RafeAldridge, 2019-01-10:
Absolutely perfect @matteocrippa 

Can read the ID of a Mifare NFC tag and it matches the ID I get in other applications. 

Really appreciate your efforts on this and hope I've been some help as a tester ;)

chunhunghan, 2019-01-18:
Have the same issue in branch master and refactor version work for me.
@matteocrippa , thank you for your amazing work.

ruan65, 2019-01-20:
Can not get it run on iPhone 7.

I am getting pod install error that I do not know how to fix:

Analyzing dependencies
Fetching podspec for `Flutter` from `.symlinks/flutter/ios`
Fetching podspec for `flutter_nfc_reader` from `.symlinks/plugins/flutter_nfc_reader/ios`
[!] Failed to load 'flutter_nfc_reader' podspec: 
[!] Invalid `flutter_nfc_reader.podspec` file: undefined method `swift_version=' for #<Pod::Specification name="flutter_nfc_reader">.

 #  from /Users/a/dev/Android/flutter_hello_rectangle/ios/.symlinks/plugins/flutter_nfc_reader/ios/flutter_nfc_reader.podspec:19
 #  -------------------------------------------
 #    s.frameworks = 'CoreNFC'
 >    s.swift_version = '4.0'
 #    s.ios.deployment_target = '8.0'
 #  -------------------------------------------

closed 2019-02-07 by matteocrippa