dotintent/flutter-nfc-reader #16: Cannot get serial number

DanielFreiburger, 2019-01-03:
At the moment it's only possible to read the data of the tag.
But what I need is the serial number.

matteocrippa, 2019-01-04:
@DanielFreiburger give a try to `feature/refactor` branch

DanielFreiburger, 2019-01-08:
@matteocrippa tried the `feature/refactor` branch but get the error
`type '_Uint8ArrayView' is not a subtype of type 'String'` 
in flutter_nfc_reader.dart:30

matteocrippa, 2019-01-08:
@DanielFreiburger did you test on iOS or Android? or pop-ups at compile time?

DanielFreiburger, 2019-01-08:
@matteocrippa I tested it on a connected android device

matteocrippa, 2019-01-08:
@DanielFreiburger prob is due to `null`, btw managed to move to empty string, let me know if it works.

DanielFreiburger, 2019-01-09:
@matteocrippa I still get this error because `id `of `NfcData `is defined as `String `but `nfcId `in the `Map `returned by `_channel.invokeMethod('NfcRead')` is of type` _Uint8ArrayView`

matteocrippa, 2019-01-09:
@DanielFreiburger it should be fixed now, sadly not having an android device makes me struggle a little and Android studio is not getting the proper linting

DanielFreiburger, 2019-01-09:
Works fine now
Thanks

closed 2019-01-09 by DanielFreiburger

ferdiebirdie, 2021-03-23:
> @DanielFreiburger give a try to `feature/refactor` branch

Is this branch still available? I can only see master and dev.