dotintent/flutter-nfc-reader #18: MIFARE Classic serial number

DanielFreiburger, 2019-01-09:
Will it be possible to get serial number of a MIFARE Classic tag as well?

matteocrippa, 2019-01-09:
No experience with that, if we found someone able to support why not.
Btw is that different from #16 ?

DanielFreiburger, 2019-01-09:
If I find something helpful, I'll pass it on to you.
Yes, it is different from #16 because the invokeMethod is not even called in this case.

Schillen, 2019-01-10:
since MIFARE is not working for now, is it possible to update the readme with the readable types?

what do you think about this article about integrating nfc on android? https://medium.com/@ssaurel/create-a-nfc-reader-application-for-android-74cf24f38a6f

matteocrippa, 2019-01-10:
@Schillen but according to #15 we can read the id of MIFARE, isn't it?

Schillen, 2019-01-10:
@matteocrippa #15 is executable, but i am not getting useful data. i only receive the "diamond question mark icon" as Identifier and Status none.

lewixlabs, 2019-01-11:
Mmmm, if you're talking about Mifare cards, maybe you need to make Android working in "card reader" mode, not NFC, to receive ATR from card contactless and if you need to send further apdus.
https://github.com/googlesamples/android-CardReader

Matteo, I work with smartcards, i wanted to make a flutter plugin to work in cardreader mode, but finding your existing project, maybe best solution is integrate this part, if you didn't do already :)

matteocrippa, 2019-01-11:
@lewixlabs any PR is always welcome 👍

Schillen, 2019-01-11:
reading ID / serial-number for Mifare is working for me as well.