dotintent/flutter-nfc-reader #20: Android: Receiving ID & Status but no content

MaxiSonntag, 2019-01-16:
Hey there,

I'm currently testing this plugin on Android devices (Android 7.0 & Android 8.0) and I'm using the [NDEF Tag Emulator](https://play.google.com/store/apps/details?id=com.maxsoft.ndeftagemulator&hl=de) as an NFC Tag, as well as a bank card for testing purposes.

By using the `feature/refacor` branch I was able to get the plugin working. I'm successfully receiving `id` and `status` in the `NfcData`-Object. However, I currently can not receive any content, as the `content` property is always empty.

Any ideas on how to solve this? Someone may test it with the mentioned app as well, maybe it's just some kind of encoding issue with the app.

Thanks in advance!

matteocrippa, 2019-02-07:
@MaxiSonntag did you test with latest stable version on master?

rotoxl, 2019-02-12:
Hello,

same problem here, I'm getting an empty value from the new NfcData object. Version 0.0.21 does the work

matteocrippa, 2019-02-12:
I think we need to debug, I have no android device here so, if any want help and use debugger to step debug that it would be great

rotoxl, 2019-02-12:
My testing has been done on iOS, I don't really know if the undesired behaviour can also be reproduced on Android.
Thanks for the quick response

matteocrippa, 2019-02-12:
Ok so better open a separate proper issue or we make a mess.

Mapk26, 2019-02-15:
Hello,

i'm testing on android device, 8.0, plugin version 0.0.23.

I receive the content but in a format with prefix "Ten" followed by the content (plain text), where Ten = T + en(english).
It seems the NDEF message is converted to string and not looking for the payload only.

Status and ID are correctly received.

matteocrippa, 2019-03-19:
up @MaxiSonntag  any update? I will close feel free to reopen if needed

closed 2019-03-19 by matteocrippa