dotintent/flutter-nfc-reader #24: Tag content is not always returned with last version 0.0.23

fvisticot, 2019-02-04:
on iOS, with the last version the content is not always returned (sometimes yes, sometimes the content is empty).

It was working fine with previous version.

Not sure it is linked but I have the error in the console:
```
[VERBOSE-2:shell.cc(186)] Dart Error: Dart_InvokeClosure expects argument 'closure' to be of type Instance.
[VERBOSE-2:shell.cc(186)] Dart Error: Unhandled exception:
NoSuchMethodError: Closure call with mismatched arguments: function 'call'
Receiver: Closure: () => void from Function '_startMicrotaskLoop@7048458': static.
Tried calling: call(Instance of '_ByteDataView')
Found: call() => void
#0      Object.noSuchMethod (dart:core/runtime/libobject_patch.dart:50:5)
[VERBOSE-2:shell.cc(186)] Dart Error: Unhandled exception:
NoSuchMethodError: Closure call with mismatched arguments: function 'call'
Receiver: Closure: () => void from Function '_startMicrotaskLoop@7048458': static.
Tried calling: call(Instance of '_ByteDataView')
```

matteocrippa, 2019-02-04:
Bit hard to debug with data few data, seems more like the NFC session has been instantiated more than a time and then fails, is it possible somehow?

CosimoSguanci, 2019-02-13:
I'm experiencing this issue too.

fvisticot, 2019-02-14:
I have forked this repo to add support for description and nfc sessions for iOS and android.
Not perfect yet but it seems to work and the issue you refer is closed as well.
Take a look on this fork and feel free to improve it (not production ready yet): 
fvisticot / flutter-nfc-reader

matteocrippa, 2019-02-14:
Feel free to make a PR

closed 2019-09-29 by matteocrippa