dotintent/flutter-nfc-reader #25: Getting an empty ID on iOS

CosimoSguanci, 2019-02-07:
Hi, I get an empty ID while reading NFC on iOS, in Android I receive the right ID.

matteocrippa, 2019-02-07:
Which kinda NFC tag are you using?

CosimoSguanci, 2019-02-07:
NXP MIFARE Ultralight - NTAG213

matteocrippa, 2019-02-07:
I think the top would be to debug using swift code and making a breakpoint to see if in the payload NFC core provides there's this value and so later work on mapping, can you do that test?

CosimoSguanci, 2019-02-07:
Searching on the internet I found that CoreNFC doesn't provide that info yet...so I guess this is the problem.

henrysachs, 2019-10-12:
This can be done in IOS 13 now and should be tackled next :)

mazensabouni, 2020-12-10:
Hello, Does anyone reached to something in this issue? It is available on NXP tag reader or TagInfo
[image]