dotintent/flutter-nfc-reader #26: Exception when repeat read

marianoarga, 2019-02-22:
I got the following exception when already read a Tag and tried to do it again:

> W/Binder  (25582): Caught a RuntimeException from the binder stub implementation.
> W/Binder  (25582): java.lang.IllegalStateException: Reply already submitted
> W/Binder  (25582): 	at io.flutter.view.FlutterNativeView$1.reply(FlutterNativeView.java:174)
> W/Binder  (25582): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler$1.success(MethodChannel.java:203)
> W/Binder  (25582): 	at it.matteocrippa.flutternfcreader.FlutterNfcReaderPlugin.onTagDiscovered(FlutterNfcReaderPlugin.kt:120)
> W/Binder  (25582): 	at android.nfc.NfcActivityManager.onTagDiscovered(NfcActivityManager.java:445)
> W/Binder  (25582): 	at android.nfc.IAppCallback$Stub.onTransact(IAppCallback.java:82)
> W/Binder  (25582): 	at android.os.Binder.execTransact(Binder.java:697)

matteocrippa, 2019-02-23:
@marianoarga you can try with new master branch version thx to @xPutnikx PR

marianoarga, 2019-02-25:
@matteocrippa Will try, thanks

matteocrippa, 2019-03-19:
@marianoarga is ok? if so feel free to close

closed 2019-03-19 by matteocrippa

marianoarga, 2019-03-19:
Didn't had any time to test, project got idle.
I will reopen if I got any issues in the future, many thanks!