dotintent/flutter-nfc-reader #29: Exception thrown when reading multiple tags

aHalyk, 2019-03-07:
E/JavaBinder(15083): *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
E/JavaBinder(15083): java.io.IOException
E/JavaBinder(15083): 	at android.nfc.tech.BasicTagTechnology.connect(BasicTagTechnology.java:85)
E/JavaBinder(15083): 	at android.nfc.tech.Ndef.connect(Ndef.java:71)
E/JavaBinder(15083): 	at it.matteocrippa.flutternfcreader.FlutterNfcReaderPlugin.onTagDiscovered(FlutterNfcReaderPlugin.kt:127)
E/JavaBinder(15083): 	at android.nfc.NfcActivityManager.onTagDiscovered(NfcActivityManager.java:445)
E/JavaBinder(15083): 	at android.nfc.IAppCallback$Stub.onTransact(IAppCallback.java:83)
E/JavaBinder(15083): 	at android.os.Binder.execTransact(Binder.java:731)

This was after holding the NFC part of the phone by a tag until it triggered this error. The amount of times before this happens is random. I have seen it throw this error after 3 reads or 50.
I tried modifying the kotlin code of the plugin surrounding the problem with a try and catch with a ndef.close() inside of the catch. I also tried putting the "on tag discovered code" onto its own thread to see if that would help but still have issues.

Things noticed during debugging
1. When plugged into the computer via USB and running in debug mode, the exception happens but the app will continue to read cards. When the app is not running while connected to a computer once the exception happens it stops reading.

aHalyk, 2019-03-08:
I changed the on tag discovered function inside of the kotlin files and it got rid of the unhandled exception. I assumed that was the issue for the tags failing to be read after continuous reads but apparently not.

Changed code for onTagDiscovered below "Sends failed ndef?.connect()" message instead of just letting it be unhandled.

  override fun onTagDiscovered(tag: Tag?) {
	Thread{
        // convert tag to NDEF tag
        val ndef = Ndef.get(tag)
        // ndef will be null if the discovered tag is not a NDEF tag
        // read NDEF message
		var id = ""
		var message = ""
		try
		{
        ndef?.connect()
         message = ndef?.ndefMessage
                          ?.toByteArray()
                          ?.toString(Charset.forName("UTF-8")) ?: ""
        //val id = tag?.id?.toString(Charset.forName("ISO-8859-1")) ?: ""
         id = bytesToHexString(tag?.id) ?: ""
		}catch(e: Exception)
		{
		message = "    failed ndef?.connect()"
		id = "-1"
		}
        ndef?.close()
        if (message != null) {
            val data = mapOf(kId to id, kContent to message, kError to "", kStatus to "read")
            eventSink?.success(data)
        }
	}.run()
    }

matteocrippa, 2019-03-08:
feel free to push a PR, it take care of review and merge @aHalyk

henrysachs, 2019-10-13:
this should be fixed now. See version 0.1.0

closed 2019-10-13 by henrysachs