dotintent/flutter-nfc-reader #30: Android: can only scan one tag

KillianByrne, 2019-03-09:
Hi there, I'm new to flutter. I tried using the sample code from example/lib/main.dart and I am currently only able to scan one tag. My question is, is this plugin capable of scanning more than one NFC tag? If so, could you explain how? 
Cheers

KillianByrne, 2019-03-09:
if it is relevant, the example StartNFC() function on the readme, doesn't work wither, the line : 
response = await FlutterNfcReader.read;
throws an error: 'A value of type 'Stream<NfcData>' can't be assigned to a variable of type 'NfcData'.'

chwiz143, 2019-03-27:
@KillianByrne It seems the response type had changed while the old documentation was unchanged.
Did you get it to work? Else try with the updated code on the [readme](https://github.com/matteocrippa/flutter-nfc-reader/blob/master/README.md).

RubenPividori, 2019-03-27:
Hi there, I'm facing the same problem.
Using the demo app, first time I execute the StartNFC() function it reads the NFC, but if I try to execute it another time, it invokes the default nfc reader app.
It seems there's nothing to listen to nfc events, like the first execution doesn't dispose everything it has to.
I tried to execute twice the reading function, and also to execute the stop function between the two reading, without luck.  
I'm using Android 8.1.0

Thank you.

leephillips1, 2019-03-30:
As a workaround for now, you can store the StreamSubscription in a variable and call cancel() on it when you stop. This will allow StartNFC() to function properly when called again.

theLee3, 2019-03-30:
As a workaround for now, you can store the StreamSubscription in a variable and call cancel() on it when you stop. This will allow StartNFC() to function properly when called again.

matteocrippa, 2019-09-29:
Still relevant? If not close

CosimoSguanci, 2019-10-07:
> As a workaround for now, you can store the StreamSubscription in a variable and call cancel() on it before you call FlutterNfcReader.stop. This will allow StartNFC() to function properly when called again.

Which StreamSubscription are you talking about?

henrysachs, 2019-10-13:
is fixed with version 0.1.0 upgrade accordingly

closed 2019-10-13 by henrysachs