dotintent/flutter-nfc-reader #32: Plugin is not able to detect any RFID tags

magnus-lpa, 2019-03-19:
I am testing the plugin on an Android device, but I am unable to scan any RFID tags. 

This is the code I'm using:

`        try {
            FlutterNfcReader.read
                .listen((data){
                    print( "--------- nfcData  content: ${data.content}" );
                })
                ..onDone((){
                    print( "--------- NFC stream done" );
                })
                ..onError((e){
                    print( "--------- NFC stream error" );
                });
        }
        on PlatformException catch (e) {
            print( "--------- PlatformException  e: ${e.toString()}" );
        }
        catch (e) {
            print( "--------- General Exception  e: ${e.toString()}" );
        }`

All that happens when I start the NFC scanning is that I get this output in the Flutter debug console:

`I/flutter (24899): lifecycle AppLifecycleState.inactive
I/flutter (24899): lifecycle AppLifecycleState.resumed`

Also, you guys seriously need to update your README.md example code both on GitHub and on Pub - it's not using streams (i.e. it's more than a month outdated).

matteocrippa, 2019-03-19:
@magnus-lpa we are open to PR, you know it's a open source project that is managed, debug and improved in spare time and for free, if you have any suggestion or improvement feel free to open a PR.

Btw which version are you using? `master` branch?

magnus-lpa, 2019-03-19:
Yes, I'm using the master branch.

I understand that it's an open source project, but as a newbie to Flutter and Dart it's very off putting to any potential user when the example code of a plugin doesn't even compile, it makes you question the quality of the plugin itself.

matteocrippa, 2019-03-19:
Sadly, as you can imagine I get nothing for this project, it's free and opensource, so my attention is focus on my job.

As soon as I will have free time, I will take care of the duty, but as usual any support is always welcome.
Otherwise it's turn to be a free help desk, more than a share and open source project, and it's far from my approach.

magnus-lpa, 2019-03-19:
I understand and I appreciate the hard work 👍 

Hopefully I can get the plugin to work, but I'm a newbie at Flutter and Android/iOS so I have no idea what could be wrong...

corjanbos, 2019-03-29:
Same problem here. Testing on a Samsung A5 2017, Android 8.0.0. Trying to get the example working. At first field values (id, content, error, status) are displaying NULL, and after scanning a public transport card, values are updated with an empty string value. So something happens, but no valid data.