dotintent/flutter-nfc-reader #37: MissingPluginException running Example on IOS

mzelles, 2019-04-26:
Hey, 

I've tried to run the example on my phone (iPhone X IOS 12.2). 
The version of flutter is 1.2.1 and dart 2.1.2.

The app starts, I can also click on the button to scan a tag. If my phone recognizes the nfc-tag, flutter throws the following error exception below. 

Hopefully someone can help me with this issue...

> flutter: NFC: Scan readed NFC tag
> flutter: ══╡ EXCEPTION CAUGHT BY SERVICES LIBRARY ╞════════════
> flutter: The following MissingPluginException was thrown while activating platform stream on channel
> flutter: it.matteocrippa.flutternfcreader.flutter_nfc_reader:
> flutter: MissingPluginException(No implementation found for method listen on channel
> flutter: it.matteocrippa.flutternfcreader.flutter_nfc_reader)
> flutter:
> flutter: When the exception was thrown, this was the stack:
> flutter: #0      MethodChannel.invokeMethod 
> package:flutter/…/services/platform_channel.dart:300
> flutter: <asynchronous suspension>
> flutter: #1      EventChannel.receiveBroadcastStream.<anonymous closure> 
> package:flutter/…/services/platform_channel.dart:490
> flutter: <asynchronous suspension>
> flutter: #30     MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart)
> flutter: <asynchronous suspension>
> flutter: #31     FlutterNfcReader.read
> package:flutter_nfc_reader/flutter_nfc_reader.dart:63
> flutter: #32     _MyAppState.startNFC
> package:flutter_nfc_reader_example/main.dart:31
> flutter: <asynchronous suspension>
> flutter: #33     _MyAppState.build.<anonymous closure>
> package:flutter_nfc_reader_example/main.dart:99
> flutter: #34     _InkResponseState._handleTap 
> package:flutter/…/material/ink_well.dart:513
> flutter: #35     _InkResponseState.build.<anonymous closure> 
> package:flutter/…/material/ink_well.dart:568
> flutter: #36     GestureRecognizer.invokeCallback 
> package:flutter/…/gestures/recognizer.dart:120
> flutter: #37     TapGestureRecognizer._checkUp 
> package:flutter/…/gestures/tap.dart:242
> flutter: #38     TapGestureRecognizer.acceptGesture 
> package:flutter/…/gestures/tap.dart:204
> flutter: #39     GestureArenaManager.sweep 
> package:flutter/…/gestures/arena.dart:156
> flutter: #40     _WidgetsFlutterBinding&BindingBase&GestureBinding.handleEvent 
> package:flutter/…/gestures/binding.dart:218
> flutter: #41     _WidgetsFlutterBinding&BindingBase&GestureBinding.dispatchEvent 
> package:flutter/…/gestures/binding.dart:192
> flutter: #42     _WidgetsFlutterBinding&BindingBase&GestureBinding._handlePointerEvent 
> package:flutter/…/gestures/binding.dart:149
> flutter: #43     _WidgetsFlutterBinding&BindingBase&GestureBinding._flushPointerEventQueue 
> package:flutter/…/gestures/binding.dart:101
> flutter: #44     _WidgetsFlutterBinding&BindingBase&GestureBinding._handlePointerDataPacket 
> package:flutter/…/gestures/binding.dart:85
> flutter: #48     _invoke1 (dart:ui/hooks.dart:223:10)
> flutter: #49     _dispatchPointerDataPacket (dart:ui/hooks.dart:144:5)
> flutter: (elided 31 frames from package dart:async)
> flutter: ═════════════════════════════════════

matteocrippa, 2019-09-29:
@mzelles which version are you using? `master` ?

henrysachs, 2019-10-13:
should be fixed with version 0.1.0

closed 2019-10-13 by henrysachs