dotintent/flutter-nfc-reader #38: Project is discontinued, looking for new mantainer

matteocrippa, 2019-07-26:
If any want to continue can add to contributor or transfer of the project so can continue to manage that in the future

huzhiren, 2019-07-29:
@matteocrippa You could let me join.

f1re, 2019-07-29:
since i am using it productive soon, i forked it and fixed already some bugs and made some improvements on my fork

matteocrippa, 2019-07-29:
@f1re if you want feel free to make a pr for that so also other can benefit of that

xiaods, 2019-08-27:
@f1re are you sure the plugin is working? i use it and can't scan any nfc tag .

xiaods, 2019-08-27:
@matteocrippa i currently working on nfc project, i can support this project.

henrysachs, 2019-10-08:
I implemented something to scan multiple tags maybe you wanna chat about it and merge it back into the lib?

matteocrippa, 2019-10-08:
hey @henrysachs, yep feel free to make a PR, so we can review with other maintainer

henrysachs, 2019-10-08:
who is the new maintainer?

matteocrippa, 2019-10-08:
still me for now, but few time to look at

henrysachs, 2019-10-08:
yeah I dont have that much time either but think on using that package for a little project myself. Maybe we can do something on par because few time and few time is a little more time. :D

I think you can assume a PR from me tomorrow is that okay for you? :)

huzhiren, 2019-10-09:
I used this package before, but it looks like not stable (only scan once and then stop working, tested on Samsung),   and is not working on ios,  then I switch to [nfc_io](https://pub.dev/packages/nfc_io), much stable so far.

henrysachs, 2019-10-09:
Hey I can understand your situation but this will change soon. Maybe you keep an open eye on the plugin. :)

huzhiren, 2019-10-09:
Yes, I understand that.  But the client don't care about that at all.

matteocrippa, 2019-10-09:
@ryanhz tbh the only one scan was the expected behaviour, btw now with help of @henrysachs we are going to add unlimited scan support.

Hope see you coming back when customer will pay you for iOS version, feature that we have at least.

ryanhz, 2019-10-09:
no, when i said only one scan, i meant it really scan once only for the lifetime of the app, the last time as I test, you need to disable nfc and re-enable nfc on the phone and restart the app to have it able to scan again.

matteocrippa, 2019-10-09:
You know at the end, open source project are always open and welcome to improvements by anyone :)