dotintent/flutter-nfc-reader #47: Can't read NFC data field (Android)

davystrong, 2019-09-18:
When reading a tag on Android (Galaxy S8), I can read the device id but the content field is empty. When I read using other apps, I get the content I expect. Any suggestions as to what I might be doing wrong? I simply use `FlutterNfcReader.read()` and wait for the Future to return.

matteocrippa, 2019-09-29:
What is the format the Tag is written?
Did you try debugging the lib and see if the kotlin code get the value?

davystrong, 2019-10-07:
Sorry for the delay. I didn't have access to the device for a while. I scanned the tag with another app that works and it says:
```
Tag type: ISO 15693; STMicroelectronics
Technologies available: NfcV, NdefFormatable
```
Is that the information you wanted?
About debugging the Kotlin code, no I didn't try that as I have no experience with plugins. Do I need any additional setup? (I have Kotlin installed)

davystrong, 2019-10-21:
So I managed to get the Kotlin debugger working. It seems that it's reading a lot of null values. In the readTag function, the tag variable has a value but ndef = Ndef.get(tag) returns null. Here's the value of tag, if it helps:
```
tag = {Tag@14196} "TAG: Tech [android.nfc.tech.NfcV, android.nfc.tech.NdefFormatable]"
mConnectedTechnology = -1
mId = {byte[8]@14209} 
mServiceHandle = 1
mTagService = {INfcTag$Stub$Proxy@14210} 
mTechExtras = {Bundle[2]@14211} 
mTechList = {int[2]@14212} 
mTechStringList = {String[2]@14213} 
shadow$_klass_ = {Class@10872} "class android.nfc.Tag"
shadow$_monitor_ = 0
```

davystrong, 2019-10-21:
It seems the problem is that the tag is reading as NdefFormatable (I didn't realise there was a difference), however, it already has a message that I would like to read. Any ideas?

davystrong, 2019-10-21:
Ok, it seems it was my misunderstanding of NFC technology that caused this problem. I assumed that Ndef V was the same as NFC V, however I must have been mistaken. If I edit the Kotlin code to read NFC V, it works correctly. I'll close the issue.

closed 2019-10-21 by davystrong