dotintent/flutter-nfc-reader #48: IOS 13 write mode

TimSchue, 2019-09-23:
Hi,
IOS 13 is now able to write on NFC. Are you planning to implement that in the near future?
p.s. great work guys

matteocrippa, 2019-09-29:
Seems relevant example can be found here: https://developer.apple.com/documentation/corenfc/creating_nfc_tags_from_your_iphone

mx4g, 2020-04-07:
> Hi,
> IOS 13 is now able to write on NFC. Are you planning to implement that in the near future?
> p.s. great work guys

Do me a favor thank you，How to Set 'Enable Capabilities / Near Field Communication Tag Reading'   ，flutter run failed

henrysachs, 2020-04-07:
[image]
You need to open the project in xcode and select the highlighted + button.

mx4g, 2020-04-07:
> [image]
> You need to open the project in xcode and select the highlighted + button.

Thanks