dotintent/flutter-nfc-reader #49: how do write information

JERINJACOBCHERIAN, 2019-09-26:
I was trying to use your plugin to send and receive JSON data using NFC in two phones.

In your example, I was able to see how to read NFC data from another device, but how can we send data using NFC from one device?

And when I tried to use 

      print(userLoad.firstName);
      FlutterNfcReader.write("path_prefix","tag content").then((response) {
        print(response.content);
      });

It threw the error
E/MethodChannel#flutter_nfc_reader(20479): Failed to handle method call
E/MethodChannel#flutter_nfc_reader(20479): kotlin.KotlinNullPointerException
E/MethodChannel#flutter_nfc_reader(20479): 	at it.matteocrippa.flutternfcreader.FlutterNfcReaderPlugin.onMethodCall(FlutterNfcReaderPlugin.kt:158)
E/MethodChannel#flutter_nfc_reader(20479): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:222)
E/MethodChannel#flutter_nfc_reader(20479): 	at io.flutter.embedding.engine.dart.DartMessenger.handleMessageFromDart(DartMessenger.java:96)
E/MethodChannel#flutter_nfc_reader(20479): 	at io.flutter.embedding.engine.FlutterJNI.handlePlatformMessage(FlutterJNI.java:656)
E/MethodChannel#flutter_nfc_reader(20479): 	at android.os.MessageQueue.nativePollOnce(Native Method)
E/MethodChannel#flutter_nfc_reader(20479): 	at android.os.MessageQueue.next(MessageQueue.java:326)
E/MethodChannel#flutter_nfc_reader(20479): 	at android.os.Looper.loop(Looper.java:160)
E/MethodChannel#flutter_nfc_reader(20479): 	at android.app.ActivityThread.main(ActivityThread.java:6669)
E/MethodChannel#flutter_nfc_reader(20479): 	at java.lang.reflect.Method.invoke(Native Method)
E/MethodChannel#flutter_nfc_reader(20479): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
E/MethodChannel#flutter_nfc_reader(20479): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
E/flutter (20479): [ERROR:flutter/lib/ui/ui_dart_state.cc(148)] Unhandled Exception: PlatformException(error, null, null)
E/flutter (20479): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
E/flutter (20479): #1      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:316:33)
E/flutter (20479): <asynchronous suspension>
E/flutter (20479): #2      FlutterNfcReader.write (package:flutter_nfc_reader/flutter_nfc_reader.dart:76:37)
E/flutter (20479): <asynchronous suspension>
E/flutter (20479): #3      _HomeMaterialState._read (package:link_me/screens/user_details_material.dart:100:24)
E/flutter (20479): <asynchronous suspension>
E/flutter (20479): #4      _HomeMaterialState.build.<anonymous closure>.<anonymous closure> (package:link_me/screens/user_details_material.dart:71:37)
E/flutter (20479): #5      _InkResponseState._handleTap (package:flutter/src/material/ink_well.dart:654:14)
E/flutter (20479): #6      _InkResponseState.build.<anonymous closure> (package:flutter/src/material/ink_well.dart:729:32)
E/flutter (20479): #7      GestureRecognizer.invokeCallback (package:flutter/src/gestures/recognizer.dart:182:24)
E/flutter (20479): #8      TapGestureRecognizer._checkUp (package:flutter/src/gestures/tap.dart:365:11)
E/flutter (20479): #9      TapGestureRecognizer.acceptGesture (package:flutter/src/gestures/tap.dart:312:7)
E/flutter (20479): #10     GestureArenaManager.sweep (package:flutter/src/gestures/arena.dart:156:27)
E/flutter (20479): #11     GestureBinding.handleEvent (package:flutter/src/gestures/binding.dart:222:20)
E/flutter (20479): #12     GestureBinding.dispatchEvent (package:flutter/src/gestures/binding.dart:198:22)
E/flutter (20479): #13     GestureBinding._handlePointerEvent (package:flutter/src/gestures/binding.dart:156:7)
E/flutter (20479): #14     GestureBinding._flushPointerEventQueue (package:flutter/src/gestures/binding.dart:102:7)
E/flutter (20479): #15     GestureBinding._handlePointerDataPacket (package:flutter/src/gestures/binding.dart:86:7)
E/flutter (20479): #16     _rootRunUnary (dart:async/zone.dart:1136:13)
E/flutter (20479): #17     _CustomZone.runUnary (dart:async/zone.dart:1029:19)
E/flutter (20479): #18     _CustomZone.runUnaryGuarded (dart:async/zone.dart:931:7)
E/flutter (20479): #19     _invoke1 (dart:ui/hooks.dart:263:10)
E/flutter (20479): #20     _dispatchPointerDataPacket (dart:ui/hooks.dart:172:5)
E/flutter (20479): 

Could someone tell me the correct way of doing it?

matteocrippa, 2019-09-29:
@JERINJACOBCHERIAN what lib version are you using?
What's the device brand / android version?

If you want give a try to `refactor` branch

JERINJACOBCHERIAN, 2019-09-30:
I am using the master branch.

Since I am using flutter, I should be able to write to all android and the new ios13 versions, right?

matteocrippa, 2019-09-30:
On iOS writing is a new feature starting from iOS 13 that was released a few weeks a go, no implementation has been added yet, there's an issue for that ( #48 )

In general using flutter doesn't make rid of all the segregation issue provided by Android during the time of the OS development, and also not all the devices are capable to do both reading and writing of NFC.

As for testing your issue, I merged refactor into master, feel free to make an update and tell me if it now works fine.

Verify also that your devices support NFC write/read.

JERINJACOBCHERIAN, 2019-09-30:
Kindly forgive if my doubts seem to simple,I am new to flutter.

Now the first issue was solved,

but now when I use   
                        FlutterNfcReader.write("path_prefix","tag content").then((response) {
                            print(response.content);

It says write is not defined

JERINJACOBCHERIAN, 2019-09-30:
Hi,

I am sorry if my doubts seem too simple,I am new to flutter.

Now I can access the read function,but the write still says that its not defined, could you tell me how to use it properly?

                          FlutterNfcReader.write("path_prefix","tag content").then((response) {
                            print(response.content);

this is the proper way yes?

matteocrippa, 2019-10-06:
It raise any error?

MoammadHamid, 2020-01-26:
hello all,
with me it doesn't raise any errors, but it simply doesn't write anything and does no reaction.
```
FlutterNfcReader.write("path_prefix","tag content").then((response) {
                        print(response.content);
```
i tried it also with replacing path_prefix with "" and other time with " ", but still the same.

edimmeduseljac, 2021-02-08:
Hello guys, 
I'm pretty new to working with NFC Tags and I just wanted to a question... 

The parameter "path_prefix" - what is that, like i know what the tag_content parameter is, but I can't figure out what I have to use as the "path_prefix". 

```
FlutterNfcReader.write(" ", path).then((response) {
                                  print(response.content);
                                });
```
This was in the example which was provided by @matteocrippa. 

Please help...