dotintent/flutter-nfc-reader #51: Error on running example

RaMb3asT, 2019-09-30:
When trying to run the example project, I get an exception that regards the plugin implementation in Kotlin:
```
e: C:\Users\stamba\flutter\packages\flutter-nfc-reader\android\src\main\
kotlin\it\matteocrippa\flutternfcreader\FlutterNfcReaderPlugin.kt: (61, 22): 
Type mismatch: inferred type is Any! but NfcManager? was expected

e: C:\Users\stamba\flutter\packages\flutter-nfc-reader\android\src\main\
kotlin\it\matteocrippa\flutternfcreader\FlutterNfcReaderPlugin.kt: (191, 17): 
Smart cast to 'MethodChannel.Result' is impossible, because 'writeResult' is 
a mutable property that could have been changed by this time

e: C:\Users\stamba\flutter\packages\flutter-nfc-reader\android\src\main\
kotlin\it\matteocrippa\flutternfcreader\FlutterNfcReaderPlugin.kt: (212, 17): 
Smart cast to 'MethodChannel.Result' is impossible, because 'readResult' is a
 mutable property that could have been changed by this time
```

matteocrippa, 2019-09-30:
@RaMb3asT did you make a `git pull` to use latest merge in master? It should be fixed since yesterday afternoon

RaMb3asT, 2019-09-30:
Yes, I am using the latest merge.

matteocrippa, 2019-09-30:
Give a try against `refactor`

RaMb3asT, 2019-09-30:
Alright, the app now runs, however when I try to read an RFID tag (NfcV tech) I get an exception:
```
E/JavaBinder( 5545): *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
E/JavaBinder( 5545): java.lang.RuntimeException: Can't create handler inside thread that has not called Looper.prepare()
E/JavaBinder( 5545): 	at android.os.Handler.<init>(Handler.java:200)
E/JavaBinder( 5545): 	at android.os.Handler.<init>(Handler.java:114)
E/JavaBinder( 5545): 	at it.matteocrippa.flutternfcreader.FlutterNfcReaderPlugin.onTagDiscovered(FlutterNfcReaderPlugin.kt:224)
E/JavaBinder( 5545): 	at android.nfc.NfcActivityManager.onTagDiscovered(NfcActivityManager.java:599)
E/JavaBinder( 5545): 	at android.nfc.IAppCallback$Stub.onTransact(IAppCallback.java:83)
E/JavaBinder( 5545): 	at android.os.Binder.execTransact(Binder.java:453)
```

mahendravikhar, 2019-10-02:
I tried by running 

`git pull origin master`

but it says Already up to date than I tried with the refactor branch and I got the message updated. but when I am running the application I am now getting message

`Package doesn't exist (the pubspec for flutter_nfc_reader 0.0.23 from git has version 0.1.0).`

kousherAlam, 2019-10-06:
# Same Problem Here 

Launching lib/main.dart on MI 5 in debug mode...
Initializing gradle...
Resolving dependencies...
Running Gradle task 'assembleDebug'...
e: /Applications/flutter/.pub-cache/git/flutter-nfc-reader-016f82008532b7d1a325d4e7a8172eca9531d35f/android/src/main/kotlin/it/matteocrippa/flutternfcreader/FlutterNfcReaderPlugin.kt: (61, 22): Type mismatch: inferred type is Any! but NfcManager? was expected
e: /Applications/flutter/.pub-cache/git/flutter-nfc-reader-016f82008532b7d1a325d4e7a8172eca9531d35f/android/src/main/kotlin/it/matteocrippa/flutternfcreader/FlutterNfcReaderPlugin.kt: (191, 17): Smart cast to 'MethodChannel.Result' is impossible, because 'writeResult' is a mutable property that could have been changed by this time
e: /Applications/flutter/.pub-cache/git/flutter-nfc-reader-016f82008532b7d1a325d4e7a8172eca9531d35f/android/src/main/kotlin/it/matteocrippa/flutternfcreader/FlutterNfcReaderPlugin.kt: (212, 17): Smart cast to 'MethodChannel.Result' is impossible, because 'readResult' is a mutable property that could have been changed by this time

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_nfc_reader:compileDebugKotlin'.
> Compilation error. See log for more details

matteocrippa, 2019-10-06:
@kousherAlam read above the suggestion please, try the `refactor` branch and verify if it works.
In general don't use the issue as a contact center, review first other reply and check if there's already a solution.

kousherAlam, 2019-10-06:
hi, @matteocrippa i tryed `refactor` branch but got the following message. 

The current Dart SDK version is 2.5.0.

Because `drinkwell` depends on flutter_nfc_reader from git which requires SDK version >=1.19.0 <2.0.0, version solving failed.

matteocrippa, 2019-10-06:
git pull and it should work

kousherAlam, 2019-10-07:
hi, @matteocrippa it works. but priviously i use FlutterNfcReader.read.listen for continus stream,  but in current read() method it does not work like this. it just fired one time. how can i fix this problem?

closed 2019-10-09 by matteocrippa