dotintent/flutter-nfc-reader #52: Compiler Errors using plugin in project

Tobisaninfo, 2019-10-09:
Hi,

I'm using your plugin in a flutter app (SKD: >=2.0.0-dev.68.0 <3.0.0) and added the plugin to the dependencies as you wrote in the readme. Then compiling the app for android, I got 3 kotlin errors.

```
e: /Users/tobias/Tools/flutter/.pub-cache/git/flutter-nfc-reader-016f82008532b7d1a325d4e7a8172eca9531d35f/android/src/main/kotlin/it/matteocrippa/flutternfcreader/FlutterNfcReaderPlugin.kt: (61, 22): Type mismatch: inferred type is Any! but NfcManager? was expected
e: /Users/tobias/Tools/flutter/.pub-cache/git/flutter-nfc-reader-016f82008532b7d1a325d4e7a8172eca9531d35f/android/src/main/kotlin/it/matteocrippa/flutternfcreader/FlutterNfcReaderPlugin.kt: (191, 17): Smart cast to 'MethodChannel.Result' is impossible, because 'writeResult' is a mutable property that could have been changed by this time
e: /Users/tobias/Tools/flutter/.pub-cache/git/flutter-nfc-reader-016f82008532b7d1a325d4e7a8172eca9531d35f/android/src/main/kotlin/it/matteocrippa/flutternfcreader/FlutterNfcReaderPlugin.kt: (212, 17): Smart cast to 'MethodChannel.Result' is impossible, because 'readResult' is a mutable property that could have been changed by this time
```

How can I fix that issue? I'm using the latest version of the flutter toolchain (1.9.1+Hotfix 4)

Thanks for help.

matteocrippa, 2019-10-09:
`refactor` is much more updated, but now I merged it to master, so give a git pull and try again

Tobisaninfo, 2019-10-09:
Thank you for your quick help. Now the plugin gets compiled without any issue.

closed 2019-10-09 by Tobisaninfo