dotintent/flutter-nfc-reader #55: update pub.dev information

henrysachs, 2019-10-09:
The Information that are displayed on the website https://pub.dev/packages/flutter_nfc_reader are old and need to be updated.

henrysachs, 2019-10-09:
with some info on how to do this i would be happy to take this task.

matteocrippa, 2019-10-09:
Yep, need to push, but the refactor was still in progress, think for now we can do that once we merge your very good PR

henrysachs, 2019-10-13:
So how do we push an update to the registry? Can I do this? I also would like to update the readme to use the pub registry instead of the git repo with a sidenote on how to get the latest features.

matteocrippa, 2019-10-13:
I should be able to do that right now, it should show the new version 0.1.x

henrysachs, 2019-10-13:
yeah! we now have a 89 scoring! https://pub.dev/packages/flutter_nfc_reader#-analysis-tab-

matteocrippa, 2019-10-13:
Yep need to figure out how to improve dart doc, and for sure ppl start using back the one from dartpub will make it more visible

matteocrippa, 2019-10-13:
I created a `development` branch, update there md file, then think I can look at dartdoc and release
Update also readme.md according, so stable will stay in master, development will keep experimental

henrysachs, 2019-10-13:
So everythings updated i think we can close this now. :)

closed 2019-10-13 by henrysachs