dotintent/flutter-nfc-reader #60: activity is null in FlutterNfcReaderPlugin

AntSwift, 2019-10-22:
I think I understand what is going on here but please forgive any inaccuracy. 

The init method of FlutterNfcReaderPlugin is accessing the activity variable without any guard against null.

In my app I am using [the workmanager plugin](https://pub.dev/packages/workmanager) which is attempting to initialise flutter in a worker thread and therefore there is no activity.

It should be a simple change to resolve this but can anyone see any broader issues which may arise?

henrysachs, 2019-10-22:
can you provide a reproduceable example? If the activity check is the only problem i dont see issues arising from this. If you feel confident provide a pr which fixes this. :)

AntSwift, 2019-10-22:
PR submitted 👍 https://github.com/matteocrippa/flutter-nfc-reader/pull/61

matteocrippa, 2019-10-26:
@AntSwift thx for the PR, wonder if not better to strip off the logs or we will have unexpected verbose messages in the apps

AntSwift, 2019-10-26:
> @AntSwift thx for the PR, wonder if not better to strip off the logs or we will have unexpected verbose messages in the apps

It wasn't my intention to include the logging changes in the PR. I will correct.

PR #63 (2019-11-19): Added null check on initialization to support background thread scenarios.
Resolved #60
closed 2019-11-19 by matteocrippa