dotintent/flutter-nfc-reader #72: Content is empty

speedlightwp, 2020-03-03:
Hi,

I have few cards, I can see only the ID however the content is always empty. How do I read the sector of the card, I presume this where I can retrieve the CAN number, expiry date and other information. Please advise.

Thanks!

crsmillan, 2020-04-15:
I have the same situation

spravesh1818, 2020-09-03:
I suppose you are checking with the bank credit cards. I came across the same situation. It seems as though the NFC libraries on flutter can only handle the ndef message formats. But the bank cards use other formats.This makes the card to be shown as empty. You can use native android applications like NFCTools to read the data in it. Though you can't see all the data you can see the technologies used.