dotintent/flutter-nfc-reader #73: Gradle Version

phillipo6, 2020-03-21:
Hi, i can't compile my app in release because this library does use grade version 4.10.1.

`A problem occurred evaluating root project 'flutter_nfc_reader'.                                                        > Failed to apply plugin [id 'com.android.library']                                                                        > Minimum supported Gradle version is 5.4.1. Current version is 4.10.1. If using the gradle wrapper, try editing the distributionUrl in E:\flutter\.pub-cache\git\flutter-nfc-reader-888b03c7158440959d660dca26ee8951c84263c6\android\gradle\wrapper\gradle-wrapper.properties to gradle-5.4.1-all.zip `

Can this be fixed?

closed 2020-03-24 by phillipo6

henrysachs, 2020-03-24:
Hi Phillpo i just saw that you closed the issue does this mean it works? because i would be glad to update the gradle version. We as maintainers just haven't had the time to do this.

phillipo6, 2020-03-24:
@henrysachs had multiple problems with some libraries because of androidx, one wasn't compatible so the gradle build process used jetifier to rebuild all dependent libraries, and so it did this for this library.
I was using gradle 5.4.1 which is why it couldn't rebuild this library. 
(But i tried to update the gradle version in a fork and it did work for me,  but was stuck at other libraries which i have now removed/replaced and i can compile release builds again)