dotintent/flutter-nfc-reader #79: Error in building example project for ios

ryanhz, 2020-05-26:
➜  Projects flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel master, 1.19.0-2.0.pre.143, on Mac OS X 10.15.4 19E287, locale en-NZ)
 
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.5)
[✓] Chrome - develop for the web
[✓] Android Studio (version 3.5)
[✓] VS Code (version 1.45.1)
[✓] Connected device (3 available)

Xcode's output:
↳
    error: Multiple commands produce
    '/Users/User/Projects/flutter-nfc-reader/example/build/ios/Debug-iphoneos/VYNFCKit/VYNFCKit.framework/Info.plist':
    1) Target 'VYNFCKit' (project 'Pods') has copy command from
    '/Users/User/Projects/flutter-nfc-reader/example/ios/Pods/VYNFCKit/VYNFCKit/Info.plist' to
    '/Users/User/Projects/flutter-nfc-reader/example/build/ios/Debug-iphoneos/VYNFCKit/VYNFCKit.framework/Info.plist'
    2) Target 'VYNFCKit' (project 'Pods') has process command with output
    '/Users/User/Projects/flutter-nfc-reader/example/build/ios/Debug-iphoneos/VYNFCKit/VYNFCKit.framework/Info.plist'

    note: Using new build system
    note: Building targets in parallel
    note: Planning build
    note: Constructing build description
    warning: The use of Swift 3 @objc inference in Swift 4 mode is deprecated. Please address deprecated @objc inference
    warnings, test your code with “Use of deprecated Swift 3 @objc inference” logging enabled, and then disable inference by
    changing the "Swift 3 @objc Inference" build setting to "Default" for the "Runner" target. (in target 'Runner' from
    project 'Runner')
    warning: duplicate output file
    '/Users/User/Projects/flutter-nfc-reader/example/build/ios/Debug-iphoneos/VYNFCKit/VYNFCKit.framework/Info.plist' on task:
    ProcessInfoPlistFile
    /Users/User/Projects/flutter-nfc-reader/example/build/ios/Debug-iphoneos/VYNFCKit/VYNFCKit.framework/Info.plist
    /Users/User/Projects/flutter-nfc-reader/example/ios/Pods/Target Support Files/VYNFCKit/VYNFCKit-Info.plist (in target
    'VYNFCKit' from project 'Pods')

Could not build the precompiled application for the device.

Error launching application on iPhone.

matteocrippa, 2020-06-16:
try to change to older build config in xcode

ryanhz, 2020-06-18:
How to  change to older build config in xcode?

matteocrippa, 2020-06-18:
https://stackoverflow.com/questions/53050108/xcode-10-how-to-switch-to-legacy-build-system

ryanhz, 2020-08-06:
I switched to legacy build system and still can't build