dotintent/flutter-nfc-reader #83: Crash android v2 embedding

cyberpwnn, 2020-11-18:
[   +1 ms] E/AndroidRuntime(24161): java.lang.RuntimeException: Unable to start activity ComponentInfo{care.myguide.app/care.myguide.app.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object
android.app.Activity.getSystemService(java.lang.String)' on a null object reference
[   +1 ms] E/AndroidRuntime(24161):     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3385)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3524)
[   +4 ms] E/AndroidRuntime(24161):     at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
[  +10 ms] E/AndroidRuntime(24161):     at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2131)
[   +1 ms] E/AndroidRuntime(24161):     at android.os.Handler.dispatchMessage(Handler.java:107)
[   +3 ms] E/AndroidRuntime(24161):     at android.os.Looper.loop(Looper.java:214)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.ActivityThread.main(ActivityThread.java:7707)
[   +2 ms] E/AndroidRuntime(24161):     at java.lang.reflect.Method.invoke(Native Method)
[   +1 ms] E/AndroidRuntime(24161):     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:516)
[   +1 ms] E/AndroidRuntime(24161):     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
[   +3 ms] E/AndroidRuntime(24161): Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object android.app.Activity.getSystemService(java.lang.String)' on a null object reference
[   +1 ms] E/AndroidRuntime(24161):     at it.matteocrippa.flutternfcreader.FlutterNfcReaderPlugin.<init>(FlutterNfcReaderPlugin.kt:61)
[   +1 ms] E/AndroidRuntime(24161):     at it.matteocrippa.flutternfcreader.FlutterNfcReaderPlugin$Companion.registerWith(FlutterNfcReaderPlugin.kt:54)
[   +1 ms] E/AndroidRuntime(24161):     at it.matteocrippa.flutternfcreader.FlutterNfcReaderPlugin.registerWith(Unknown Source:2)
[   +1 ms] E/AndroidRuntime(24161):     at io.flutter.plugins.GeneratedPluginRegistrant.registerWith(GeneratedPluginRegistrant.java:21)
[   +2 ms] E/AndroidRuntime(24161):     at care.myguide.app.MainActivity.configureFlutterEngine(MainActivity.java:30)
[   +1 ms] E/AndroidRuntime(24161):     at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onAttach(FlutterActivityAndFragmentDelegate.java:179)
[   +1 ms] E/AndroidRuntime(24161):     at io.flutter.embedding.android.FlutterActivity.onCreate(FlutterActivity.java:410)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.Activity.performCreate(Activity.java:7825)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.Activity.performCreate(Activity.java:7814)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1325)
[   +1 ms] E/AndroidRuntime(24161):     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3360)
[

TerehinAV, 2021-11-07:
I have a similar issue. 

```
The plugin `flutter_nfc_reader` uses a deprecated version of the Android embedding.
To avoid unexpected runtime failures, or future build failures, try to see if this plugin supports the Android V2 embedding. Otherwise, consider removing it since a future release of Flutter will remove these deprecated APIs.
If you are plugin author, take a look at the docs for migrating the plugin to the V2 embedding: https://flutter.dev/go/android-plugin-migration.

```

michelaSpreghini, 2021-11-08:
Same problem

matteocrippa, 2021-11-08:
Feel free to submit a PR if you have any opportunity, actually I have not that much time to take care of this project not using flutter anymore

TerehinAV, 2021-11-10:
> Feel free to submit a PR if you have any opportunity, actually I have not that much time to take care of this project not using flutter anymore

I'v figured out that you've already migrated to v2 embedding at master branch, can you please publish this version to pub.dev?